spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #83: Where are the Arduino .ino files located?

FreekvandenHeuvel, 2022-04-30:
Hello,

About a year ago, my daughter built 2 v2 detectors for a school project. Never finished the detectors due to "soldering mistakes".
Reworked them last week and the electronics and SiPM-module are correct and working now.

However can't find the necessary Arduino .ino files (apart from the Naming.ino and OLED.ino and SDCard.ino files).
Found the v1 files, but not the v2's.

Where can I find them ??

Regards, Freek

spenceraxani, 2022-04-30:
Hi Freek,

The Arduino code is located in the Arduino/ folder:
https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/tree/master/Arduino <https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/tree/master/Arduino>

The Naming.ino script is uploaded to name the detector. Then you can upload either the OLED.ino or the SDCard.ino file, depending on whether you want to run the OLED or SDCard. The reason why they are currently separated is due to the limited RAM on the Arduino. 

Thanks,

Spencer

FreekvandenHeuvel, 2022-04-30:
Hi Spencer,

Thanks for your quick response!

I feel stupid. I was looking for a file with a name like "Detector.ino" or something like that. Thought the other files, like "Naming.ino", were parts of a central file. I should have opened one of them first; then I could have seen it.

Sorry to have bothered you for something so obvious ...

Thanks, Freek

closed 2022-04-30 by FreekvandenHeuvel