spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #84: Forbush decrease

PaulHearn92, 2022-05-10:
Has anyone observed the Forbush decrease while observing Cosmic radiation (muons).  Does the Cosmic Watch detector have sufficient sensitivity to see this.

(https://www.sciencedirect.com/topics/earth-and-planetary-sciences/forbush-decrease)  

Paul Hearn

spenceraxani, 2022-05-10:
Hi Paul,

No one has yet to observe it, but we’ve been trying with an array of 10x detectors for some time. Our problem is that we haven’t had a very large CME while the detector array was running. We’re looking for it in Boston, although it would be easier to see at higher altitudes and latitudes. 

Thanks,

Spencer

PaulHearn92, 2022-05-10:
Thanks for swift reply.  Maybe I will be the first.  I’m now monitoring 24/7

 

Best wishes,

 

Paul.

 

Paul Hearn

Radio Astronomy Section Director

UKRAA Trustee

 

¦¦¦  British Astronomical Association  ¦¦ 

Burlington house, Piccadilly, W1J 0DU

 <https://britastro.org/section_front/24> https://britastro.org/section_front/24

 <https://www.youtube.com/user/britishastronomical/playlists> https://www.youtube.com/user/britishastronomical/playlists

   

e      ***@***.***> ***@***.***

a     RG6 1BU UK

t     +44 (0)7967 388 578