spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #85: OLED not starting up when plugged in

JMulder99, 2022-05-16:
My OLED screen won't start up when the Arduino is plugged in. I already tried:
1. Checking the OLED.ino file reads "1" to turn on the OLED.
2. Replacing the screen
3. Switching the VCC and GND pins using 4 male to female wires. 
4. Checking the voltage across the VCC (4.6V), and SCL and SDA (around 3.2V)

The DC-DC booster circuit isn't working, but I don't think this is needed for the screen to turn on. What is left to troubleshoot?

spenceraxani, 2022-05-16:
Hi, 

The OLED requires that the VCC pin is one of the edge pins. You shouldn’t switch the VCC and GND pins — it may be that you fried something sending the VCC to GND. There’s not much to the OLED screen, the 4 connections go directly to the Arduino. If the Arduino has the code uploaded, the screen should turn on. Some of the screens I’ve purchased have noticeable cracks in the screen, primarily in the upper corners. If the screen is cracked it will not work. A smaller percentage of screens, maybe 1% or so, I’ve found that the ribbon cable on the back of the screen was soldered crooked. 

The DC-DC booster takes the 5V and boosts it to 29.6V. I you have the correct input voltage (4.6V is fine), then the problem lies in one of the components in the DC-DC booster circuit. The most common issue I see is the soldering on the LT3461 pins. I would check the orientation and add some new solder. If the problem persists after checking the LT3461 and adding new solder to all the components (including the resistors), then the likely issue is either that the LT3461 needs to be replaced or the larger inductor needs to be replaced. Analog.com <http://analog.com/> will send you two free LT3461 if you have an account. I’ve had to replace maybe 1% of the LT3461s, and sub<1% of the time the larger inductor.

I would aim to get the OLED working first. When you swapped the VCC and GND pin, there’s a chance you fried the Arduino. If you cannot get the screen working, I would be cautious working on the rest of the circuit, since the Arduino is nearly impossible to replace.

Thanks,

Spencer

JMulder99, 2022-05-18:
Thank you @spenceraxani for your response. I got in contact with an electrician and he resoldered the OLED socket. It fixed the issue.

closed 2022-05-18 by JMulder99