spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #87: OLED.ino versus SDcard.ino loaded on Arduino Nano.

JMulder99, 2022-05-21:
Hi,
I read the instruction manual and found the following:
> Either the OLED.ino or the SDcard.ino code can be uploaded to the Arduino Nano.
Does this mean that if I want the screen to fuction, the SDcard won't function? Or vica versa, if I want data to be written to my SDcard, will the screen not function?

I noticed both programs used >50% storage during sketching.

spenceraxani, 2022-05-21:
Hi,

Yes, the libraries for running the OLED and the SDCard are both too large to fit into the Arduino’s memory at the same time.

Thanks,

Spencer

jgOhYeah, 2022-05-21:
Hello,
A few years ago I managed to get some firmware working using smaller libraries that supported both the display and the sd card at the same time. See [here](https://github.com/jgOhYeah/NonOfficial-StuffForCosmicWatchMuonDetectors) if interested. I get the feeling that it is never far from a stack crash and I no longer have access to the actual hardware to test with, so use at your own risk :).