spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #88: Efficiency measurements or estimates

jeroenvantilburg, 2022-05-29:
Hi there,

I have been wondering about the actual efficiency of detector for measuring a single muon, since I could not find any numbers on that. Has anyone performed such a measurement or are there some estimates on the efficiency? Presumably the efficiency depends on the muon path length and the distance between the muon path and the SiPM. It would also be interesting to know an estimate of the peak height (in mV) per detected photon in the SiPM. Does anyone know this number? With that information it might be possible to use Poissonian statistics to derive some efficiency number.

As a first and preliminary attempt I sandwiched 3 detectors: one master and two slaves. When a muon passes through the top and the bottom detector, I expect that it must have gone through the middle one. Hence, I can measure the efficiency of the middle detector. I found the coincidences between the two slaves (middle and bottom) using this [tool](https://jeroenvantilburg.nl/cosmicwatch/). With an hour of data taking the middle detector measures 117 hits out of the 163 that went through the top and the bottom detector (and have a peak height > 50 mV). This gives an efficiency of 72+-4 % which is averaged over the full surface in this flat orientation. Does this look reasonable? There are probably a few things that can be improved in this setup. In any case, I plan to make more efficiency measurements (longer, different orientations, different detectors, shielding) in the future. Let me know if you have suggestions.

spenceraxani, 2022-05-29:
Hi,

Good question — I’ve had the efficiency measurement on my back burner for quite a while. A 

I’ve tried pulling out single photons to verify the gain curve as a function of over voltage, but have so far been unable to get the noise sufficiently low to measure it. I’m much more familiar with single photon detection with PMTs, where the integral of the single photon waveform divided by a load resistance of 50Ohms at gain of 10E7 gives a peak voltage of ~5mV. I was hoping my intuition would be sufficient for the SiPMs as well, although I think I’m missing something when applying this logic to the FAST and SLOW output of the SiPM. Anyways, hope to directly figure this one out in the future is a low noise power supply.

A group at the Departamento de Ciencias made a full Geant4 simulation of the detector to characterize the detectors 2019. Their paper can be found here: https://arxiv.org/pdf/1903.04562.pdf <https://arxiv.org/pdf/1903.04562.pdf>. The efficiency they found was roughly ~76%. I was actually expecting it to be much higher, in the 90s. If you removed the condition “>50mV” does your efficiency go up? Judging by the attached figure from the PhysicsPaper.pdf, it really looks like we’re not missing many events — that is, the coincident events are well above the low energy threshold. I bet we’re missing some corner clippers.

It should also be noted that the OLED screen itself causes about 5% dead time, so I’m guessing 8 out of 163 events happened while the screen on the middle detector was updating.

Thanks,

Spencer

jeroenvantilburg, 2022-05-30:
Oops, I did not realize that the OLED screen eats so much deadtime. Before I used to measure only in the SDCard mode, observed that the deadtime was negligible, and forgot about it. Looking again at the data files I now see that the deadtime is 18.6%. I completely overlooked this. This should explain why the efficiency was so low.

I have measured again the efficiency using the exact same setup, but know with the SD card code loaded into the Arduino. The measured deadtime according to Arduino code is now negligible (only 0.27%). Now (in slightly more that an hour) the middle detector sees 362 hits out of a total of 376 that went through both the top and the bottom. This gives an efficiency of (96.3 +- 1.0)%. Much closer to your expectations. This is without any requirements on the bottom detector. When I require that the hits in the bottom detector > 50 mV, the efficiency goes up to 326 / 336 = (97.0 +- 0.9)% (keep in mind that the requirement is on the trigger source; not the middle detector). If I compare this with the efficiency obtained with the OLED screen on, there is still some small gap when I correct for the dead time. Maybe the dead time calculation for the OLED code is a bit too small? In any case will make further efficiency measurements (with the SDCard code) in the near future.

It would still be good to know the peak voltage per photon. I do not really understand how you want to measure it (also I know little about PMTs). I thought of changing the REF voltage of the Arduino to some small number (say 100 mV) and hope to see the individual photon peaks in the (ADC of mV) histogram. This is how I am used to measure this number. However, the REF pin is already hardwired in the PCB, so this means modifying the board and/or cutting Arduino headers, which would not be so easy for me...