spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #93: Plans to release v3?

osullivryan, 2022-08-28:
Hello! Awesome project!

I notice that you have some videos that go over the v3 detector, but the links to the GitHub project have been removed. Do you have any plans to release the designs for the v3 detector? 

Thanks!

Ryan

spenceraxani, 2022-08-29:
Hi Ryan,

Negative on the v3 release. That version was a bit too difficult to build by hand and would have required much more troubleshooting help, which I would be unable to provide. However, I’m planning on releasing a similar version in the near future that will have many of the same features as v3.

Thanks,

Spencer

─
Dr. Spencer N. Axani
***@***.*** 

Assistant Professor
The University of Delaware
208 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

osullivryan, 2022-08-29:
Hello Spencer,

Understood! Thanks for the quick reply. When I go to build this I'll just grab the latest plans. 

Thanks again,

Ryan

closed 2022-08-29 by osullivryan