spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #95: Position of SiPM resistor R12

Phoenix1747, 2022-09-29:
Hi, I was wondering why you chose to place the SiPM resistor R12 on the SiPM daughter board and not close to the first op amp stage and R4. Is this a deliberate choice or is it just a convenient place for this part?

Thanks!

spenceraxani, 2022-09-30:
Hi,

I choose to place it near the SiPM because that would allow you to have the signal output through the BNC connection without needed to populate the MainPCB. Although, in hindsight, I think I would have fully removed R12 to reduce the pulse attenuation going into the first amplifier. Right now the amplifier has to work pretty hard to amplify the signal.

Feel free to let me know if this doesn’t fully answer your question.

Thanks,

Spencer
─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
208 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

Phoenix1747, 2022-09-30:
> Although, in hindsight, I think I would have fully removed R12 to reduce the pulse attenuation going into the first amplifier. Right now the amplifier has to work pretty hard to amplify the signal.

Wouldn't fully removing R12 result in no current flow from the SiPM and therefore no signal? Or do you actually mean R4? I'm not sure how it would attenuate the signal either, to be honest.

spenceraxani, 2022-09-30:
It essentially has to do with terminating the line to match the impedance. When you terminate a 50Ohm line with a 50Ohm resistor, you pass the maximum amount of power, but the voltage gets cut in half (it acts as a voltage divider). If you do not terminate the line, you don’t get that 1/2 attenuation, but you end up with your signal reflecting. If the line is short, say from the SiPM to the first op amp, the reflection doesn't really matter, but when the line is long, say from the BNC to an oscilloscope, it does matter. 

I sort of misspoke in my last message, I wouldn’t fully remove R12, instead I would put a ~1-10k resistor in its place. This gives the current a path to ground, but it also ~minimizes the attenuation of the signal (half attenuation of the signal to ~1-5% attenuation). Since the path from the SiPM to the first op amp is relatively short, there wouldn’t be significant reflections on this line. However, when you look at the signal on an oscilloscope, you would be forced to terminate the line at the oscilloscope.

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
208 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

Phoenix1747, 2022-09-30:
Oh, I see now! So it has to do more with the BNC connector and the issues occuring due to external connections and not the SiPM/amplifier directly. Thanks!

closed 2022-10-01 by Phoenix1747