spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #96: Low count rates

Kristalek, 2022-11-01:
Hi, 

I have a problem with low count rates, after I turn it on, only 1 false count will appear. Im sending photos. Do you have any suggestion where could be the problem?

I built 2 main desks and 1 SiPM (doesnt work). I also checked if there are used right componets.

Have a nice day :)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

spenceraxani, 2022-11-01:
Hi Kristalek,

Thanks for the photos, it looks like you did a good job building the detector. Hard to point out the exact component that is causing the problem, but given the images, the issue is likely one of the following:
	1. Most likely culprit: poor solder connection on one or more components in the amplifying/peak detection circuit. I would hit these components with new solder. Suspect connections are R4, R5, and R6. Very much R12 on the SiPM PCB.
	2. The solder connection on the SiPM looks OK, and I know that you tested the resistance in both directions, however, it’s a bit tough to say how good that connection is once you interfaced it with the scintillator.

I also noticed that you do not have optical gel on the SiPM. While this is not the cause of your problem, you will find that you get a low rate as fewer photons make it to the SiPM. 

Thanks,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
208 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

Kristalek, 2022-11-01:
Hi, 

I tried to reheat and add more solder. Than I measured resistances and it equals of I think right values(because there are some resistance divisors, from scheme). 

And nothing changed.

Do you have any suggestions? 

Thanks for your time :)

spenceraxani, 2022-11-01:
Just to clarify, you are saying that you added new solder to every connection in the circuit? Including the resistors that I pointed to? 

If you have a light-leak into the scintillator, that could also cause your problem. Are you sure that you taped the SiPM scintillator combo sufficiently well?

If so, and you are confident in your solder job on the SiPM, we can use an oscilloscope to narrow down the location of the problem. Since you have the correct voltage going to the SiPM, you should be able to connect the BNC output of the detector to an oscilloscope and search for the SiPM pulses. They will be ~10-100mV in height and ~250ns in length. If you see them, the issue must be in the amplifying or peak detector circuit. If not, the issue is with the SiPM PCB.

Thanks,

Spencer
─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
208 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

Kristalek, 2022-11-01:
Hi, 

Yes I added new solder to connections including connections of resistors you mentioned. 

Hmm. Are you saying that: If the scintillator isn't enought tight to the SiPM, it won't work, right?

Next time I'll be working on it, I'll try to look at the pulses.

Thanks for help.

spenceraxani, 2022-11-01:
The detector works by looking for small flashes of light from charged particles passing through the scintillator. If there is a light-leak, even the tiniest  one, into the scintillator it’ll saturate the detector and you wouldn’t see any pulses.

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
208 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

Kristalek, 2022-11-03:
The problem was in a light leak.

Thank you for your time you spent solving problems. :)