spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #99: I accidentally ordered V1's PCB and parts.

ilkaydemirhan, 2023-01-20:
Hi,

I wanted to build Muon-Detector-v2.

In the process, I downloaded the purchasing list of both V1 and V2 to my computer. I also downloaded the Gerber files of both V1 and V2 as well. 
While ordering the PCB, I accidentally ordered the PCB of V1. Then I ordered the parts for V1. So, I messed up the files.

This happened completely by accident. I caused this error myself. Because the PCB and purchasing list files of V1 and V2 had the same name and I mixed them up. And all orders have arrived. 
Now I have two options. The first is to reorder the V2's PCB and parts (This is going to be expensive). The second option is to make V1. 
If V1 works fine and is not very different from V2, I think I should do V1. 

I'm asking you, which one would you recommend me between these two options?

Best Regards,

Ilkay

spenceraxani, 2023-01-20:
Hi IIkay

I would definitely go with V2. V2 has a lot of additional features that makes it a significant upgrade (coincidence connection and sdCard primarily). I’m happy to send you two V2 PCBs if you like though.

Most of the components between v1 and v2 are the same. So, you shouldn’t have to order too many new parts. 

Thanks,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

ilkaydemirhan, 2023-01-21:
Hi Spencer,

Thank you for your kind response. I will follow your advice and make V2. 

Thank you for your kind suggestion. I like too much.

Best Regards,

Ilkay

ilkaydemirhan, 2023-02-01:
Hi Spencer,

I will reorder PCB and missing parts. I wanted to check compatibility of purchasing list and Gerber file.

1. PCB_Gerber_files_wLargerSnapOffConnectors.zip
2. Updated parts list last year.

I will order 1 and 2 if they are compatible with each other.

I would be glad if you comment on this.

Thanks,

Ilkay.

spenceraxani, 2023-02-01:
Hi IIkay,

If the files are contained in the same repository (v2), then the combination of 1 and 2 will work. 

Thanks,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

ilkaydemirhan, 2023-02-01:
Hi Spencer,

Yes both are from the same repository but there are two Gerber files. One was changed 5 years ago, the other was changed last year.

I guess I should use the Gerber file modified last year.

Thanks,

Ilkay

spenceraxani, 2023-02-01:
All the files are actually fine. The newest file was generated with larger snap-off connections since a company in Europe required them. The electronics part list was only updated to include new link for components that were out of stock due to the global chip shortage.

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

ilkaydemirhan, 2023-02-01:
Hi Spencer,

I understand.  Thank you for the information.

Best Regards,

Ilkay.