spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #100: 2 Different kind of PCB

Miriamxan, 2023-01-25:
Hi Spencer, 

I builded a detector before with some different components, for example AD8604 instead of  LT1807. It worked for a while and then it has stopped working. Then  I ordered all the components just the same with yours. Any of them (I tried to build 6 detectors ) is not working, there is no signal. It is just triggered and then nothing... 
So I m trying to understand the reason, I took a voltage measurement over each element. There is no electric on R8, R16 and R15. But I can get the 29.50 V on Sipm connector. I wonder if u can say something on this situation. 

Also , I printed two different types of pcb unconciously . One of them has some "NS" footprints that u write on the smt populating list. Other one doesnt have that NS component s footprints. So no need to take care.
I used the first one that has not any footprint belong to "NS" components. So what do you think about which one ı should use? 

Thanks, 
Meryem

spenceraxani, 2023-01-26:
Hi Meryem,

Thanks for the email. The AD8604 is not suitable for this application. The bandwidth and slew rate are much too low for the input signal. There are likely other issues with it as well, but those two alone make it insufficient. 

You’ll likely have to do some troubleshooting to identify the problem. Since you confirmed that you measure the correct voltage going to the SiPM (~29V), that means that you are able to test to see if you populated the SiPM board correctly. After interfacing the SiPM to scintillator, and taping it up, you can plug the BNC output into an oscilloscope. If your SiPM PCB was properly built, you should see the SiPM pulses. They are 10s of mV in height and 100s of ns in width, triggering the detector at the Hz rate. If you do not observe these, that means that a problem lies in the SiPM PCB, otherwise, you can move on to see if that signal get’s amplified at test point 2 (TP2).

NS refers to “No stuff” which means not to populate that component. Those board should be fine, if they came from the v2 repository.

Thanks,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

Miriamxan, 2023-01-28:
Hi again Spencer, 

Thanks a lot for your answer. Yeah, I actually was pretty sure about AD8604 but we couldnt find LT1807 that time so I wanted to use another TIA opamp. It worked for a while or we thought ıt worked :)  
Whatever, I builded 5 detectors. I also followed their SiPM signal on oscilloscope as u said. I couldnt get any signal like you described. It is obvious there is a problem on SiPM Pcb. But I'm also skeptical about PCBs in general. So I prepared an excel file that shows Voltages of some important components for each of them. If you see something that ı cant understand maybe it can help me. So also these are some informations about my detectors ; 
- I couldnt use any optic jel between SiPM and Scintillator , still waiting for order 
- 2 detectors has 1k Ferrite Bead not 2.5 k. I m not sure if it s that much important or not
 Except these , all the same as yours.  Sorry to ask so many questions but  I m so interested in this project because I will have the opportunity to visit many cities in the world in the coming period, I want to use the detector in the places I go. Thanks a lot. 
[5 detectors.xlsx](https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/files/10528401/5.detectors.xlsx)

spenceraxani, 2023-01-30:
Hi,

Since you get 29.5V on the HV connection to the SiPM board and do not see pulses out the BNC connection, that localizes the problem (or one of the problems) to the SiPM PCB. The most common issue on this board is the orientation of the SiPM or a poor solder connection on the SiPM. Feel free to send me a picture of your boards — I might be able to identify the problem. It is a bit surprising that you cannot extract pulses through all 5 of the detectors, are you confident that you are using your oscilloscope correctly?

The ferrite bead doesn’t really matter, any value would be fine. The lack of optical gel would reduce the amount of light you see by a significant amount, which means that many events fall below your trigger threshold and do not trigger the detector. But, even without it, you should still see ~0.5Hz or so.

Thanks,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

Miriamxan, 2023-01-30:
Hey, 

I attached some pics of detectors. So you can see all populated components and their direction. Also I tested SiPM for both direction.  You can see also Oscilloscope and my style of using:) Tbh I used to use manual oscilloscope but this one eaiser to use so I hope I m using in correct way. 
Thank you very much in advance for all your help really.

All the best, 
Meryem

[CosmicWatch.Meryem.zip](https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/files/10539799/CosmicWatch.Meryem.zip)

spenceraxani, 2023-01-31:
Hi Meryem,

Thanks for the video and images. I can see that the scintillator is not fully light-tight — there are regions around the corners that aren’t fully taped up which would prevent the detector from functioning. The scintillator needs to be fully light-tight. 

Thanks,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

ilkaydemirhan, 2023-02-01:
Hi Meryem,

I'm living in Ankara. Where do you live? 

Where did you get the parts?  Especially SIPM and scintillator?

Thanks,

Ilkay.

Miriamxan, 2023-02-01:
Hi İlkay, 

I m in İstanbul, and working on CosmicWatch for a project. We bought all components from abroad, different countries. But ı can send the companies who can get these components. You can reach me on instagram if u use. 
(lafenetre_demarie) 

Thanks, 
Meryem