spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #101: LT1807 Replacement

alex-stendahl, 2023-02-02:
Hi,

Currently the LT1807 is out of stock so is it possible to replace it with for example LTC6227HMS8E? What parameters should one look at to see if a amp is a suitable replacement for LT1807, I saw at least that slew rate and bandwidth are important, and im assuming the voltage supply span. Are there others?

Best,
Alex

spenceraxani, 2023-02-02:
Hi Alex,

It’s hard to say without trying it. There are many many different ways that would prevent a direct substitution for another op amp. The thing that stands out to me with the LTC6227 is that it says:

"The LTC6226/LTC6227 are optimized for high bandwidth applications, and have not been designed to directly drive capacitive loads.”

The peak detector is a very large capacitive load (10nF). So this may be a deal breaker. The output swing low is also about twice as high as the LT1807, which means that you may not see the smaller pulses.

You need an op amp that includes the negative rail on the input, and get’s as close to the negative rail on the output. With sufficient bandwidth and slew rate, that can operate on single supply, that can also drive capacitive loads. A different input offset voltage could also cause problems. That being said, if you find an op amp with those characteristics, the component values around the op amp would likely need to be changed, for the feedback and gain resistors may need to be selected differently. 

I personally tried substituting tens of different op amps directly, without success. It’s my fault though — I should have designed the circuit differently. I didn’t anticipate the LT1807 being out of stock.

Thanks,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

alex-stendahl, 2023-02-02:
Hi,

Then I would also say it would not work as a replacement. Maybe one could try but as a working-out-of-the-box its not suitable. 

My followup question is that the part LT1807CMS8 should be identical to the correct part except a lot smaller roiht? So it should work normally if one adapts the footprint somehow?

Best,
Alex

spenceraxani, 2023-02-02:
I think if you could adapt the form factor, it should work.

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

closed 2023-02-02 by alex-stendahl