spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #103: Arduino sketch not  working  !! please help

musketsmoke, 2023-02-08:
anyone have a working sketch as i keep getting the same error, Compilation error: Adafruit_SSD1306.h: No such file or directory

i  am  so close to finishing and failing at this stage,  i am a hardware guy cant do this software stuff i am lame lol  

can someone please just upload a working sketch for the cosmic watch v2 that works in one file  not 3 to  join together, been trying to sort this for days and days one error  is  killing me or where and how  to fix it much appreciated
Please help

spenceraxani, 2023-02-09:
Hi,

The error is stating that you do not have the Adafruit_SSD1306 library installed. You’ll need to go to Sketch->Include Libraries->Manage Libraries, search for Adafruit_SSD1306, and install Version 1.0.1.

You’ll then receive the same error for the other libraries as well. But you can just follow the same procedure. The information is listed in the first few lines of code.

Thanks,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

musketsmoke, 2023-02-09:
Hello Dr Spencer

sorry to  bother you, i am  having huge trouble  with these sketches.
No matter what i do trying different loading  of what you said i should do i keep getting  errors.
Loading the Naming.ino alone there is no problem, putting the OLED sketch under the naming sketch i get loop errors
i have tried loading all 3 more errors and i different orders.
What  am i doing wrong? IS there  a file that i can just upload that is  stitched together so all i  have to do is add the library's  and it works ?  there are more than one wire and  spi file which one ? 
if i am having trouble with  this software i am so scared when am i if i am  going  to get it working let  alone  setting it up if it works i  am so useless with software 
i  have already  made investments in most components for 10 to be made as  i am in the UK its quite expensive to import the required things, here is my email if its easier to send a file please clit3100@yahoo.com
thank you for helping  me and  sorry to bother you with my issue.

spenceraxani, 2023-02-10:
Hi,

The naming.ino file writes the chosen name of your detector to long term memory (non-volatile memory called EEPROM). After uploading the Naming sketch, your detector permanently remembers that name. 

After that, you can upload the OLED.ino or SDCard.ino sketch. You do not need to put the OLED code “under” the naming sketch. Your detector will not forget the name that it was given using the Naming.ino sketch.

Feel free to let me know if that doesn’t solve your issue.

Thanks,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

musketsmoke, 2023-02-11:
Hello Dr Spencer

if  i understand  i  only upload the naming sketch too the nano, nothing  else
i am building the motherboards  and came across the reference NS for  R10-R14-R18-R19-R25-U7
i have  never seen the reference on any other diagram before can you clarify this for me please NS

Thank you

spenceraxani, 2023-02-12:
Hi,

Once you upload Naming.ino, it stamps the name that you chose into the Arduino’s long term memory. After that, you can upload OLED or SDCard. 

NS refers to No-Stuff, which means that you should just ignore those footprints.

Thanks,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426