spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #107: Cannot run detector_server.py or import_data_py3.py

fbarradas, 2023-07-20:
This is what happens if I try to run detector_server.py  (from Spyder from MacOS 13.4.1):
[image]

and  if I try tro run  import_data_py3.py:
[image]

fbarradas, 2023-07-20:
And this is the same working from the Linux terminal
[image]

spenceraxani, 2023-07-20:
Hi,

Where did the detector_server.py code come from? In the Arduino folder there is only import_data.py.

It looks like the first error is just that the code has spaces instead of indents, and your text editor cares about that. You would have to change the spaces into tabs, or tabs into spaces.

The second error relating to threads. I cannot test it at the moment. But I see that we are importing something called _thread, whereas the text uses something called thread. My guess is that you should be able to change thread->_thread and it will work.

Best,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

anthfuller, 2023-07-26:
I'm also seeing this error code, running "import_data_py3.py" as well. When running the script, there are five choices, the last being [h] for help. When selecting option "1", Record data on computer, the script prompts for a USB Port number, entering the number, I get this error: 
Traceback (most recent call last):
  File "/home/radiomuon/muon/CosmicWatch-Desktop-Muon-Detector-v2/Recording_Data/import_data_py3.py", line 195, in <module>
    port_name_list.append(str(port_list[int(ArduinoPort[i])-1]))
IndexError: list index out of range

Can you help with this, looks like an issue with the script on line 195?

spenceraxani, 2023-08-01:
Hi,

Would you be able to post a screen shot? Does your computer see the Arduino? As in, were you able to upload the code to the Arduino through the Arduino IDE?

Thanks,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

JG-QuarkNet, 2024-07-02:
@anthfuller Your "IndexError: list index out of range" error is different from the previous errors in this thead and likely has a separate solution.  Fortunately I had the exact same problem, which I traced to this issue: https://github.com/arduino/help-center-content/issues/155

The Linux BRLTTY utility for braille displays is sometimes enabled by default, and has been in particular for recent versions of Ubuntu.  Unfortunately, BRLTTY is programmed to take over serial ports automatically, which interferes with the `import_data_py3.py` script's scan of your ports.

You can confirm whether or not this is your problem by checking `dmesg` output related to tty and looking for a line that cites 'brltty' like the following:

```
$ sudo dmesg | grep tty 
[ 5339.669621] usb 3-4: usbfs: interface 0 claimed by ftdi_sio while 'brltty' sets config #1
```

If you don't need the `brltty` utility, the easiest solution is to uninstall it with `$ sudo apt remove brltty` (for Debian/Ubuntu, adapt for your distribution as necessary).

If you do need the `brltty` utility, you'll have to find another way to make it let go of the ports.  There's more info on that Arduino [GitHub issue](https://github.com/arduino/help-center-content/issues/155) I posted above.