spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #108: Troubleshooting

0ort9, 2023-07-20:
Hello Spencer,

building my third Watch right now and slowly getting a routine. :) I am still in love with the project! Thank you so much.

I have finished everything but did not get any detections. I tested the voltage on the main PCB which is fine. I tested the SiPM pulse with the BNC and tp1 and it looks good, but I do not get a signal on tp2.

So something between tp1 and tp2 is not working. Could you help me verify which parts could be responsible? Is it only the Amplifying circuit?

Thank you in advance and have a nice day

0ort9, 2023-07-20:
tp2 shows a signal now but tp 3 does not yet :D 

Doe the 1807 brakes fast with too much heat? It looks like the side is a little melted. Might have to replace it

spenceraxani, 2023-07-20:
Hi,

Looks like you have a problem with the op amp, the LT1807. It should not get hot. TP2 is the amplified version of TP1, via the op amp. Since you saw TP1, the problem is somewhere in the op amp or the amplifying part of the circuit. You may have either put it on backwards, or shorted something. I would replace it, double check the orientation, and make sure the legs aren’t bridged.

Best,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426