spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #110: Troubleshooting again

0ort9, 2023-09-04:
Hello Spencer,
I could need some input from your side for one of our non-working main PCBs. TP1 and TP2 show great signals, TP3 shows some weird reversed peak: 
[image]
We measured every component after TP2. Is there a way to check if the LT1807 is still working?

spenceraxani, 2023-09-05:
Hi,

That’s a new one. I can’t say for sure. Were TP1 and TP2 positive pulses? If TP2 is amplifying the signal, I think it’s a good indication that the LT1807 is working fine. For TP3, you cannot terminate the line to 50Ohms at the input to the scope, otherwise, that additional path to group will cause the pulse to decay away too quickly. You aren’t inverting the pulse on the oscilloscope are you? My recommendation would simply be to reassess your soldering and the components on the peak detector circuit. It’s going to be one of those components causing the problem.

Best,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

spenceraxani, 2023-09-05:
> Thank you for the quick response. If i compare TP1 and TP2's peaks I see an amplification of 30 instead of 10, maybe this has to do with the amp circuit after all

Is TP1 by chance terminated to 50Ohms and TP2 not? That would cause a factor of 2x difference between the two.

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426