spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #113: Coincidence confusion

WRJOLLY, 2024-02-07:
I am trying to setup two Cosmic watches in coincidence mode where can I find the coincidence.ino code mentioned in the instructions PDF? or could I just use the SD.ino code for both?
thank you

spenceraxani, 2024-02-08:
Hi,

Using either the OLED.ino or SDCard.ino code will boot them up in coincidence mode, if they are connected via the 3.5mm cable.

Best,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

WRJOLLY, 2024-02-23:
Coincidence detection works great, but I would like to get coincidence detection with roughly 8 to 9 cm space between each scintillator. Is there a way to change the 30us delay between the master and slave scintillators in the code?

spenceraxani, 2024-02-23:
Hi,

Sorry I don’t think I quite follow. You still will get coincidence detection when they are spaced some distance apart, it’s just that the rate will decrease as the distance increases. The muons are traveling relativistically, and will pass through both detector within approximately 0.3ns (when 10cm apart).

Best,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426