spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #115: I have an issue taking data from  two detectors

Omadre69, 2024-03-19:
I have two detectors and i want to run the import.py application to collect data from both arduinos.
Lets say that they have designated com ports 6 and 7.
However when it comes to selecting ports
[1] COM port 6
[2] COM port 7

 when select: 1,2
 the import._data .exe shuts down instantly
[image]

jeroenvantilburg, 2024-03-19:
Rather than the python script I would suggest to use my webapp instead:CosmicWatch readoutjeroenvantilburg.nl

Troubleshooting.jpg (view on web)

spenceraxani, 2024-04-01:
Hi,

That’s a bit strange of a problem. Are you able to record from a single detector? And, are you sure that those are your two detectors? Often [1] is some bluetooth connection. 

You could try adding some print statements to the code to see where it is shutting down. Do your detectors have names? I noticed that if you don’t provide a name via the naming.ino code, you get some non-ascii characters which could potentially cause an issue.

Best,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426