spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #116: Concerns with linked parts not matching description in purchasing list

gimmeAll, 2024-03-26:
I am just asking if the linked digikey parts are ok to buy even though the descriptions in the purchasing list do not match the linked part's description: the ferrite bead being 1kohm, yet stated as 2.5kohm on the list; the .1 uf and 1uf ceramic capacitor's temperature coefficient being Y5V, yet stated as X7R; the 	Linear Voltage Regulator being 250ma, yet stated as 300ma. That is all I have found so far, yet there might be more. I know it is probably not an issue, I just wanted to ask.

spenceraxani, 2024-04-01:
Hi,

All those are fine and won’ t affect the functionality of the detector. Thanks for the notes though.

Best,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

dedwid, 2025-02-05:
Hi, I know it probably won't affect the circuit as well, but I've also found another unmatched description: In digikey's description, the 20nF capacitor has a voltage of 100V instead of the 50V stated on the purchasing list. Could this difference lead to an issue on the circuit? I just want to make sure before purchasing all the components :)
Thanks in advance.

spenceraxani, 2025-02-05:
Hi,

Thanks for the question. That won’t have an impact. That’s the maximum voltage allowed for the capacitor. The actual voltage on the capacitor is only a few volts.

Best,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426