spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #119: Trouble finding the coincidence.ino code

SophieRAllen, 2024-08-14:
Hi,
I am doing this project as part of an EPQ at school and am quite new to programming. However, I can’t seem to find the coincidence code to upload into the IDE. Where can I find it and do I have to install any other libraries aside from the CH340g and those listed on the README file?
Thanks,
Sophie

spenceraxani, 2024-08-14:
Hi Sophie,

The Arduino Code is located in the Arduino folder. You can either upload the OLED or SD Card code — the Arduino isn’t powerful enough to run both at once.

In either of the pieces of code, the first few lines tell you which libraries are needed to be installed. 

Specifically, you need to install:

  Requirements: Sketch->Include->Manage Libraries:
  1. Adafruit SSD1306     -- by Adafruit Version 1.0.1
  2. Adafruit GFX Library -- by Adafruit Version 1.0.2
  3. TimerOne             -- by Jesse Tane et al. Version 1.1.0

Best,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

hunnicho, 2024-09-16:
Hi Spencer,

I am also struggling with this issue. I cannot seem to find the coincidence.ino code in the Arduino folder. 

I'm able to see the naming, OLED, and SD Card code and have installed the correct libraries, however the coincidence.ino file that the condensed instructions reference does not seem to be available. 

Is it in a different location or was it possibly renamed?

Thank you,

Hope

spenceraxani, 2024-09-17:
Hi Hope,

Sorry, that might be a typo. Both the OLED and SD Card.ino code have the coincidence stuff built into it. 

Best,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426