spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #122: See pulses on all TPs, but screen doesn't update counts

hswhhe-uiuc, 2024-11-20:
Hi, I have a question about the Arduino/screen readouts. I looked at the BNC signal and all the test points on my detector, and I can see reasonable signals from the Oscilloscope. However, the screen counts doesn't update at all, except one fake count at the beginning. I tried to lower the signal threshold from Arduino but it doesn't change anything. Do you have any suggestions? The pictures below are the signal from BNC&TP2, BNC&TP3, and my main board. Thank you so much!

[image]
[image]
[image]
[image]

spenceraxani, 2024-11-20:
Hi,

I would recommend changing the trigger threshold back to its original value. I think your issue is simply the solder connection on the Arduino. I can see several suspect connections. I would also solder all the pins.

Best,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

hswhhe-uiuc, 2024-11-21:
Hi Spencer,

Thank you so much! It worked after more soldering on Arduino.

closed 2024-11-21 by hswhhe-uiuc