spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #124: Silk screen not made at SiPM pcb during production

dedwid, 2025-02-15:
Hi,
I made an order to Elecrow with [PCB_Gerber_files_wLargerSnapOffConnectors.zip], and they have recently contacted me due to a problem with the silk screen located where the SiPM is soldered. I asked them about it, and they told me this:

“The characters are placed on top of the solder mask, and the software will automatically remove the character layer covered by the solder mask exposed copper during production. So there is no silk screen printing when making according to the file.”

They sent me this pictures as well:

[image]

[image]

Will the circuit work without this silk screen? If not, I thought about sending them the [PCB_Gerber_files.zip] to see if there's any difference.

Thank you for all your work in this project!

Kind Regards,
David

wffinney, 2025-03-04:
I too have had these boards printed, and soldered my SiPM following the location shown for Pin 1 in the image in the instructions.  I do not get the behavior shown in the video where there is a very high or "open" resistance when probed in one direction and a measurable resistance when measured by swapping the leads.  Is this a known behavior with this board?

wffinney, 2025-03-04:
> I too have had these boards printed, and soldered my SiPM following the location shown for Pin 1 in the image in the instructions. I do not get the behavior shown in the video where there is a very [...]

Oh... I think I answered my own question with the information about the meter setting in issue 81
https://github.com/spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2/issues/81#issuecomment-1118913551