spenceraxani/CosmicWatch-Desktop-Muon-Detector-v2 #125: How data is coded in text data file

athornett, 2025-03-08:
Hi Spencer,
Ted Cline and I are trying to produce some routines in Easy Radio Astronomy to plot CosmicWatch Muon Detector data.
Can I ask you how coincidence detections are recorded in the data files and distinguished from non-coincidence measurements? Or is it just that once two detectors are in coincidence mode then all data from that moment on is conincidence data?
Andy
Dr Andrew Thornett
SARA & BAA-RAG
www.astronomy.me.uk
Lichfield Radio Observatory, UK

spenceraxani, 2025-03-09:
> athornett
>  created an issue

Hi Andy,

For this particular version of the detector, the detector that was placed in coincidence mode will only save coincident events. So all triggers on the coincident detector are coincident. In the next version of the detector, all events will be saved with a flag indicating which ones were coincident. 

Best,

Spencer

─
Dr. Spencer N. Axani
***@***.*** ***@***.***> 

Assistant Professor
The University of Delaware
203 Sharp Lab
104 The Green, Newark, DE 19716
(608) 572-8426

athornett, 2025-03-09:
Thanks!

Does that mean that V3 is not yet available? If so how did you create the plots that show coincidence vs non coincidence detections?

Andy